gdombiak/OctoPod #449: OctoPod and Apache reverse proxy?

DariBer, 2021-01-24:
Has anyone here got OctoPod on iOS got to work when the OctoPrint is behind a apache reverse proxy?
If anyone have a working config, please post it here.

gdombiak, 2021-01-24:
Hi @DariBer, I have OctoPrint running behind nginx (used as reverse proxy). The idea would be similar. You will need to have a proxy for the website and a proxy for each webcam that you have installed. If can share my setup if you want to go the nginx route. For Apache as a reverse proxy you can follow [this](https://www.digitalocean.com/community/tutorials/how-to-use-apache-http-server-as-reverse-proxy-using-mod_proxy-extension) and [this](https://httpd.apache.org/docs/2.4/howto/reverse_proxy.html) guides

DariBer, 2021-01-24:
Thanx for your sender @gdombiak, I have an apache and nginx as reverse proxy. So if you can share your config it would be great.

gdombiak, 2021-01-24:
Here you go. A content like this is in /etc/nginx/sites-available/default. I terminate HTTPS on the reverse proxy and use [letsencrypt](https://letsencrypt.org/) for my SSL certs.

```
# Server for octopi.local
server {
    listen       5555;
    server_name  myserver.com;
    ssl    on;
    ssl_certificate    /etc/letsencrypt/live/myserver.com/fullchain.pem;
    ssl_certificate_key    /etc/letsencrypt/live/myserver.com/privkey.pem;

    location / {
        proxy_pass http://192.168.1.55/;
        proxy_set_header Host $http_host;
        proxy_set_header Upgrade $http_upgrade;
        proxy_set_header Connection "upgrade";
        proxy_set_header X-Real-IP $remote_addr;
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
        proxy_set_header X-Scheme $scheme;
        proxy_http_version 1.1;

        auth_basic           "Protect Octopi Console";
        auth_basic_user_file /etc/apache2/.htpasswd;

        client_max_body_size 0;
    }

    location /webcam/ {
        proxy_pass http://192.168.1.55:8080/;

        auth_basic           "Protect Octopi Console";
        auth_basic_user_file /etc/apache2/.htpasswd;
    }

    location /webcam2/ {
        proxy_pass http://192.168.1.55:8081/;

        auth_basic           "Protect Octopi Console";
        auth_basic_user_file /etc/apache2/.htpasswd;
    }

    # redirect server error pages to the static page /50x.html
    error_page   500 502 503 504  /50x.html;
    location = /50x.html {
        root   html;
    }
}
```

DariBer, 2021-01-25:
Thanx for your config :-)
Will try it later tonight.

DariBer, 2021-01-26:
@gdombiak I sort got it to work, but there's one thing I don't understand, the /webcam/ config, you use port 8080 and port 8081 for another webcam. I wonder where do you configure these ports on the OctoPrint?

gdombiak, 2021-02-07:
Pot 8080 is what OctoPi uses out of the box. I created a second service that runs mjpg_streamer in port 8081 for my second camera that is hooked up to the same RPi.

nauni77, 2022-03-30:
Hello @gdombiak , 
thanks for your configuration! The infos about your `proxy_set_header` settings helped me to solve my issues. Before OctoPod didn't show all values.

This configuration works well for me. :-)

Best regards, Oli