gdombiak/OctoPod #450: Camera Controls Tab

GoodyGizmos, 2021-01-26:
I'd like to be able to control the webcam's settings using OctoPod. OctoLapse implements this fairly well (scroll down to the "Better Camera Controls" section on this page). This is separate (but maybe a first step?) from OctoLapse integration into OctoPod down the road.

gdombiak, 2021-01-26:
Hi @GoodyGizmos,

Thanks for this request. It seems like OctoLapse is a very popular plugin. Been getting more request to add support for this plugin. It is currently scheduled for release 3.12. It seems to have a lot of functionality so it will take me some time to implement it.

Regards,
Gaston

GitHubGoody, 2022-01-26:
@gdombiak Since I originally posted this request, this plug-in (https://plugins.octoprint.org/plugins/camerasettings/) came out. I'm definitely interested in having this level of control over a connected webcam, but don't think OctoLapse integration is necessary. My $0.02. 😀