gdombiak/OctoPod #451: BL Touch Commands in Octopod

bkarr123, 2021-01-29:
Would there be a way to add the BL Touch commands (i.e. Probe Up/Down) into the Move tab of Octopod on iOS? I see a bunch of commands, but BL Touch commands would be great since I have to test probe my needle since it will crash into the bed if I don't do that.

gdombiak, 2021-01-29:
Thanks for sharing this idea. I'll see what I can do. I'm not aware of a way to detect if a printer has bl touch or not (one of my printer has it while the other don't) so ideally this new feature should only appear if it has bl touch.

Thanks,
Gaston

bkarr123, 2021-01-30:
[image]

Thanks! These are the commands I was looking for, the row of BL Touch commands at the bottom. Those actually populate in my Octoprint control because I also have the BL Touch plugin on my Octoprint instance.

commit 5338c462 (2021-02-07): 1. MoveSubViewController are enabled/disabled based on printer status and 2. Added support for BLTouch plugin

Closes #451
Closes #462
closed 2021-02-07 by gdombiak