gdombiak/OctoPod #453: Webcam upside down on AppleTV

mattadata, 2021-01-31:
My webcam image is only upside down on AppleTV. In OctoPod iPhone app, the image is fine. In the regular OctoPrint web interface, the image is fine.

gdombiak, 2021-01-31:
Hi @mattadata ,

Thanks for reporting this issue. I did the following and it worked for me (though I found another bug)
1. Go to OctoPrint settings and flipped the camera
2. Camera looks ok in OctoPrint (as you said)
3. Camera **initially** looks incorrect in Apple TV
4. closed the Apple TV app and opened it again
5. Camera looks ok in Apple TV

Screenshot step 1:
[image]
Screenshot step 2:
[image]
Screenshot step 5:
[image]

I will see how I can fix step 3 since that is not working as desired.

Did you follow similar steps? If not, could you give this a try? Apple TV should respect whatever settings you have in OctoPrint. Currently you need to open the Apple TV app twice for the modification to take effect. After that it works fine

mattadata, 2021-01-31:
Thanks. That solved it. I just had to open the AppleTV app a few times and then it showed the correct video.

closed 2021-01-31 by mattadata

gdombiak, 2021-02-01:
Next release will just work without having to close the app. BTW, since I was working on this part of the app .... I added more improvements to make the experience more seamless. Will open new tickets for the improvements.