gdombiak/OctoPod #461: Print job doesn’t show status

albeedelacruz, 2021-02-03:
I have the Octoprint address set as https://my.domain.com
I can see the camera and all other functions except the status.

The domain is signed via Lets Encrypt.

Everything works fine if I use the local http domain but I need to access this outside my network.

Help?

albeedelacruz, 2021-02-03:
What is weird is that it still sends a notification to my phone that the print is 50% complete.

joelstahre, 2021-06-18:
I have the exact same issue, my own domain, signed , I see camera feed and temperature. But not status of print. Sometimes I get a pop up in app saying “Not Updating”

albeedelacruz, 2021-06-19:
I don’t think anyone will respond at all. I ended up just setting an OVPN server on my Synology and access Octoprint via the local address.

BlaydeRunner08, 2021-06-19:
I have my own domain as well, octoprint.mydomain.com, but I don’t run https…. Ive had no issues…