gdombiak/OctoPod #463: Plugin Octolight

jhonny811, 2021-02-10:
Congratulations on the great app. I would like to ask you if it is possible to implement the Octolight plugin, as PSU control is already present.
Thanks a lot and congratulations again

gdombiak, 2021-02-10:
Thanks @jhonny811 for filing this nice improvement. I checked the plugin and I think it will need to be improved so that OctoPod can render the current state. I will check it in more details later.

jhonny811, 2021-02-18:
Thanks for the reply. I wanted to report a new plugin (GPIO Control). If you can integrate it, it looks really great. good job

gdombiak, 2021-02-18:
Thanks @jhonny811 for this request. I filed it in this other ticket https://github.com/gdombiak/OctoPod/issues/467. I checked this other plugin and it has proper APIs so it should be doable. I scheduled this for the next release.