gdombiak/OctoPod #464: shortcuts

huskymitch, 2021-02-12:
could you add "Print Finished" to shortcuts?

gdombiak, 2021-02-12:
Thanks @huskymitch for the feature request. Scheduled it for the next release.

Regards,
Gaston

gdombiak, 2021-06-26:
Hi @Ma3er,

Starting to work on this request. Want to confirm I got it right. Are you looking for a way to ask Siri if the print has finished? So the answer would be yes or no?

Thanks,
Gaston

Ma3er, 2021-06-26:
Hi Gaston
I thought it would be good if you could use it in a shortcut, if print
is finished then you could turn of a smart plug. so then the print is
finished that would be a trigger.
I'm not sure if this is posable.

thanks,
Dwayne

gdombiak, 2021-06-26:
Gotcha. I do not think that Siri can do what you are asking. Action starts by user saying something rather than monitoring for a condition (e.g. print finished) so it can run an action.

Having said that, the way to solve this is from OctoPrint itself. There are many plugins that could help with this depending on your setup. [OctoPrint plugin for OctoPod](https://plugins.octoprint.org/plugins/octopod/) supports IFTT that can help with this. [OctoPrint-Tasmota](https://github.com/jneilliii/OctoPrint-Tasmota) can also do this if you use tasmota switches. [OctoPrint-TPLinkSmartplug](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug) can do this if you use TPLink switches. I think there are many more plugins like those as well.

Would that help?
Gaston

closed 2021-06-26 by gdombiak

Ma3er, 2021-06-26:
Ok I’ll give that a try

Thank you