gdombiak/OctoPod #471: Apple TV

deharren, 2021-03-07:
I’m getting a “connecting to printers screen” but nothing is appearing.  Any advice on how to resolve this?  Thanks.

AudioBr4in, 2021-03-11:
Apologies if this is against regulation, but just wanted to comment that I'm encountering the same problem. Some details about my setup:

- Apple TV 4K with two different Apple IDs signed in (mine and one other)
- iCloud Sync is enabled for both OctoPod on my phone and on TV, with my Apple ID being active when opening the TV app
- Doubt it's causing any issues, but I'm also running a Pi-hole on my network. All my devices sync, and the TV app still doesn't work when I pause blocking, but figured I'd mention just in case

DadOf6, 2021-04-01:
I found myself recently with this same problem.  My setup is like AudioBr4in’s though I don’t have a pi-hole. I have tried re-syncing iCloud and still got the same message. My workaround was to remove the app and reinstall, which worked. Though obviously this doesn’t give the “why”.

deharren, 2021-04-01:
Turns out I resolved it. I was manually backing up my phone to the computer and not subscribed to iCloud. I have since subscribed for backup purposes and it fixed it!

gdombiak, 2021-08-29:
The other day my Apple TV had the same issue. The only way I was able to make the printers appear again was to go to OctoPod app in my iPhone, go to Settings -> Printers and save again each printer. As I saved them, they started to appear again on my Apple TV. Hope this trick works for you

tommy077, 2024-09-02:
> The other day my Apple TV had the same issue. The only way I was able to make the printers appear again was to go to OctoPod app in my iPhone, go to Settings -> Printers and save again each printer. As I saved them, they started to appear again on my Apple TV. Hope this trick works for you

Thanks, that did the trick. Went to the printers on my iPhone. Altered my login so I could save again. And suddenly the app on my Apple TV started to display my printer.

johnwc, 2025-03-22:
I had this issue as well, fixed it by going to iPhone app and going to Settings -> Devices -> Reset iCloud data from printers, then closing the app in Apple TV and opening again.

gdombiak, 2025-03-22:
Hi @johnwc , thanks for sharing that  workaround. I do not know why iCloud loses the information so your recommendation restores printers data in iCloud. This will make AppleTV happy again. :D 

Regards,
Gaston