gdombiak/OctoPod #475: Spool manager integration

luco85, 2021-03-12:
Hi! I use the spool manager plugin but is not seen in octopod. Is possible to integrate it in to the app? Thanks

BlaydeRunner08, 2021-03-21:
Hey @luco85, are you talking about the FilamentManager plugin?  If you are, it is supported.  If you swipe over to the third page on the "Panel" tab, you'll see the relevant information.

That being said, I have asked @gdombiak in #472 to add a bit more functionality for the plugin.

luco85, 2021-03-22:
Hi! Yes i know filament manager, i used it with octopod, but i mean about spool manager, another plugin from same developer

samwiseg0, 2021-04-25:
I too would like to see this integration. Here is a link to the source repo for reference https://github.com/OllisGit/OctoPrint-SpoolManager

kadaan, 2023-11-21:
Would someone want to try to get this into PR form and submitted to this repo?  I don't have time.

https://github.com/gdombiak/OctoPod/commit/0a91f173c0986b758ae1dba7cedf13021c45a51c

gdombiak, 2023-11-22:
Hi @kadaan ,

Thanks again for yet another contribution. My feedback here is similar to your other contribution:
1. Could you export the changes somehow so I can process them easier?
2. What are the changes in the data model? (eg. anything else besides adding 'spoolManagerInstalled' to Printer?)
3. What's new in the UI? (I assume a VC to let people select their spool)
4. Want to confirm whether you added support for alerting user when there is not enough filament in the spool before print start.

Thanks again,
Gaston

kadaan, 2023-11-22:
1. I will try to find time of XMas
2. Correct, just `spoolManagerInstalled` was added to `Printer`
3. Basically a copy of the Filament UI with some slight changes, one of which is the showing the code from SpoolManager.  The code is useful because it can help you ensure you pick the spool the corresponds with the UI
4. I did not

kadaan, 2023-11-22:
@gdombiak #672

gdombiak, 2023-11-27:
Thanks @kadaan . This PR and your other PR have now been merged (using other PRs)

closed 2023-11-27 by gdombiak