gdombiak/OctoPod #476: Implement OctoRelay Settings

lkrieger00, 2021-03-15:
First off all, i love octopod, and in my opinion it is the best App for managing Octoprint on my Smartphone. 
Just a few days ago a new Plugin called "OctoRelay Settings" was released and i really love it, because it finally allows me to controll the lights of my Printer via Smartphone so i can turn it on if I want to see the camera feed and leave it off the rest of the time. Can you please also implement this option in Octopod? I would love to use this!!

Thanks for your amazing work!

bastienstefani, 2021-09-16:
Hello guys, would love to have this too...
Currently i control my printer with octorelay, 1 relay for the motherboard so i can turn it down on print finish and one for the lights i can switch on/off when i want to check with the camera.
That would be awesome to have to control also on the phone app !

gdombiak, 2021-10-15:
Thanks @bastienstefani . I just looked at the code of that plugin and I do not see an API that returns list of relays with their labels and status. I only see an API to read the status of a relay or change it. Not efficient to read status of all relays on a single API call and no label would be displayed. :(

Once these other 2 APIs exists, I can see of adding support for this plugin.

Gaston

gdombiak, 2021-10-17:
Hey @bastienstefani ,

How about the following?

Request
```json
{
  "command": "listStatus"
}
```

Response
```json
[
  {
    "pin": 1,
    "name": "Light",
    "on": false
  },
  {
    "pin": 2,
    "name": "Printer",
    "on": true
  }
]
```

Regards,
Gaston

bastienstefani, 2021-10-17:
@gdombiak 
I can change the response structure no problem, but the pin identifier is useless as it’s the relay identifier (r1, r2…) that you have to send in the update to toggle on/off. So might be interesting to change the ‘pin’:2 to ‘relay’:’r2’. Is that ok for you ?

bastienstefani, 2021-10-17:
> @bastienstefani, the reason I added a field named **pin** with an int value is because the [_update_ and _getStatus_ commands](https://github.com/borisbu/OctoRelay/blob/29b442adb1203a5ffafe38c2e9df9a22265191e1/octoprint_octorelay/__init__.py#L211) expect a property called **pin** that is the pin number. Clients would just take whatever value was returned by _listStatus_ and use it for these other 2 APIs. What do you think?
> 
> Regards, Gaston

I made the test yesterday, it’s called pin but it’s the rX (r1, r2…) that it takes, It’s a mistake in the property naming I guess

gdombiak, 2021-10-17:
I did not test this plugin but look at this code
```python
    def on_api_command(self, command, data):
        self._logger.debug("on_api_command {}, some_parameter is {}".format(command,data))
        index = data['pin']

        settings = self.get_settings_defaults()[index]
        settings.update(self._settings.get([index]))
```
index is used for accessing an array. It seems to expect an int. Passing a string should make the API crash.

Gaston

gdombiak, 2021-10-17:
I see. That’s an internal method used by OctoPrint. The APIs exposed by this plugin seem to not use that method. I’m trying to be consistent with existing APIs. Might want to check with plugin developer on their new API. I’m just giving feedback. 

Regards
Gaston

closed 2022-01-01 by gdombiak