gdombiak/OctoPod #477: “Connect to iCloud...” message, nothing else

mwilgar, 2021-03-26:
Just installed the AppleTV app and all I get is the message "Connect to iCloud to retrieve list of printers from your iPad/iPhone". I have 2 accounts the AppleTV, the same two that are on my iPhone and iPad, I get the same result regardless of which account is active on the AppleTV. I verified that iCloud is enabled on iPhone and iPad, and octopod works fine on those.

thetwig187, 2021-04-26:
Has anyone else encountered this? I’m also only getting the “connect to iCloud” message and I have confirmed that I’m connected to iCloud both on my phone and on the Apple TV but still only get the “connect to  iCloud” message

gdombiak, 2021-08-29:
The other day my Apple TV had the same issue. The only way I was able to make the printers appear again was to go to OctoPod app in my iPhone, go to Settings -> Printers and save again each printer. As I saved them, they started to appear again on my Apple TV. Hope this trick works for you