gdombiak/OctoPod #481: Printer dashboard scroll bug

deharren, 2021-04-07:
First off great app!!  I wanted to report a hug that if you have enough multiple printers in “Printer Dashboard” view then the bottom one that extends off the screen in portrait mode will not all you to view the very bottom. 

It you scroll it down It starts to quickly climb up the screen and cut off the bottom few lines of the detailed results.  Landscape mode works fine though.  See attached video.  Thanks!!

tideline3d, 2021-07-19:
+1 on this one, annoying on the non-home button iOS devices as the swipe up gets in the way.

gdombiak, 2021-07-19:
Found the root cause. Testing before adding it for next release. Thanks guys for pushing to get this bug fixed. I thought it was above my skills to get it. :D

tideline3d, 2021-07-19:
Sweet!  I was considering just adding a dummy printer to the bottom of my list to get around it 👍

commit 3db7a410 (2021-07-20): Fixed annoying auto-scrolling of dashboard

Closes #481
closed 2021-07-20 by gdombiak