gdombiak/OctoPod #482: I am not able to delete printers from the app. Can not add printers from outside local sources.

psdavidson812, 2021-04-07:
Hi, I am not able to delete printers from the app. Can not add printers from outside local sources. ip and port number is good because I can open it in my browser. Thanks

gdombiak, 2021-08-29:
Hey @psdavidson812 , to delete a printer go to Settings -> Printers and on the printer you want to delete do a slide to the left. You will see a red button for deleting the printer. You can add printers outside of your LAN via OctoEverywhere, VPN or port forwarding + reverse proxy for minimal security.

closed 2021-08-29 by gdombiak

psdavidson812, 2021-08-29:
Nope it does not give an option and will not slide. iPhone 12

gdombiak, 2021-08-29:
What do you think of this idea of seeing a lock when the app is locked and you try to add a new printer or save changes of an existing one?

[image]

[image]

psdavidson812, 2021-08-29:
I really think that would be helpful. It would also give a visual in case
you wasn’t sure what was going on. You could say hey it must be locked let
me look somewhere in the settings.