gdombiak/OctoPod #486: Tasmota: Separate relays on the same Tasmota device show the same status

melyux, 2021-05-02:
Was happy to see the Tasmota plugin support in the app. Placement could be more convenient, but it works. One bug though is that two relays in Tasmota that use the same device show the identical state as one another despite having different states. For example, in this screenshot, one plug is on and the other is off, but they both show as on.

[image]

gdombiak, 2021-08-29:
hey @melyux , which version of tasmota plugin are you using? What happens if you pull down to force a refresh? Is the OctoPrint web UI showing a different state?

melyux, 2021-09-23:
Hey @gdombiak

I'm on version 1.1.0 of the Tasmota plug-in. Nothing happens if I pull down to refresh. The web UI shows the two plugs having different states, which is right. Only the OctoPod app shows them both on even if only one of them is on.