gdombiak/OctoPod #487: Connection Failed: Service is down or incorrect port

hugoender, 2021-05-03:
I can connect to 2 out of 3 Prusa printers. The third, gives me the "Connection Failed Service is down or incorrect port".

If in the Octopod app I go to Terminal > OctoPrint Web, I can log in just fine and the Octoprint GUI loads up fine. Both the RPi in questions and my iPhone are connected to the same network.

I've tried:

- deleting and re-adding the printer multiple times
- deleting and re-adding Application Key (I'm using QR code to scan it in)
- I've checked my router firewall and have no rules blocking anything from this device (I have no issue accessing the Octoprint GUI from my phone via web browser or the in app browser.

Any help with this would be greatly appreciated.

Here's the System Information of the Octoprint instance in question:
browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:88.0) Gecko/20100101 Firefox/88.0
connectivity.connection_check: 1.1.1.1:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1200
env.hardware.ram: 917016576
env.os.bits: 32
env.os.id: linux
env.os.platform: linux2
env.plugins.pi_support.model: Raspberry Pi 3 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.15.1
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 9.0.3
env.python.version: 2.7.13
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.6.0
printer.firmware: Prusa-Firmware 3.8.1 based on Marlin
systeminfo.generator: systemapi

tideline3d, 2021-07-19:
Are you using a .local address for this connection or an IP address?  I'd recommend trying the IP directly as sometimes .local name resolution can be finicky.

gdombiak, 2021-08-29:
any update on this issue @hugoender ?

hugoender, 2021-08-29:
I have since changed my network setup and placed all the RPi's on the same VLAN and it's working fine. Not sure what was it about separate VLANs that didn't allow it to work since the firewall didn't have any rules preventing it (that I know of) but at this point it's not an issue anymore. Thank you.

closed 2021-08-29 by hugoender

gdombiak, 2021-08-29:
Thanks @hugoender for the update!