gdombiak/OctoPod #488: Feature Request : Support Moonraker / Klipper as well

TravisWilder, 2021-05-05:
If possible it would be great if the app could Klipper Printers as well.
The Klipper can be connected with Moonraker, which as a good client documentation:

https://moonraker.readthedocs.io/en/latest/web_api/

Thanks for considering

ReubenBTalbott, 2021-09-16:
I'd love this feature as well. Currently the only thing holding me back from Mainsail is a iOS app, OctoPrint while awesome is a lot heavier than Mainsail plus I like the design of the web UI better.

psiberfunk, 2021-11-23:
Also interested in this , would be great to have octopod support klipper

gdombiak, 2021-11-23:
Hey guys,

Thanks for this cool suggestion. Unfortunately, I do not use or have Klipper so not planning on adding support for it. I took a quick look at Moonraker (never heard of it before) and even though it offers an API it seems to not be compatible with OctoPrint. 

Having said this, if anyone wants to add support for it ... I'd be glad to review changes and add them if it makes sense. Maybe a new client for Moonraker needs to be an entire new project?

Gaston

TheLion, 2023-03-18:
Octopod seems partly to work with Moonraker, when the Octoprint compatibility is enabled in Moonraker. 
Octopod shows the current temps and printer status, but nothing more and it does not refresh. See screenshot for more info. 

[image]

gdombiak, 2023-03-19:
Hi @TheLion,

Based on the information you shared, it seems like Klipper+Moonraker have compatibility with REST API but they **do not implement** webservices. Without webservices you won't get updates as the print progresses. OctoPod first makes a REST API call to get basic data and display it while it establishes a websocket connection to get stream of updates as they happen.

If you are running in the local network then it seems that websocket compatibility is not there yet. If you are running behind a reverse proxy then maybe this is a configuration of the reverse proxy.

Gaston

TheLion, 2023-03-19:
> Hi @TheLion,
>
> Based on the information you shared, it seems like Klipper+Moonraker have compatibility with REST API but they **do not implement** webservices. Without webservices you won't get [...]

Hi Gaston,

It is probably right what you are describing. Octoprint compatibility in Moonraker is limited and I doubt it will get expanded in the (near) future. 

I'm not expecting Octopod to full support Klipper/Moonraker, but it would be nice if it somehow can detect it is Klipper/Moonraker, give a different notification as the current error and refresh the temps and print status.

I do think the information for Octopod is available, looking at other plugins/programs, only different than with Octoprint and I can fully understand if you don't want to go that route.

TravisWilder, 2023-03-25:
Hi .. moonraker offers the WS and this is also used by the Webguis like mainsail
https://moonraker.readthedocs.io/en/latest/web_api/#websocket-connections