gdombiak/OctoPod #491: Enclosure Temp Control - Enclosure Plugin

Kiwistylebuild, 2021-05-19:
I have the Enclosure plugin (4.13.1) and Octopod (3.11) is showing the temp/humidity values on the first page and then the LED on/off control the third screen over.  However, I cannot find a control for the enclosure temp feature.  

I had a good look through the other issues and requests which talk about showing the temp values, and controlling the GPIOs around versions 4.2ish but nothing specific on enclosure temp adjustment.  Was this a feature that should already be working, or is it possible to have it added?

Thanks!
Philip

Kiwistylebuild, 2021-05-19:
That is awesome thanks Gaston.  Yeah main reason I (and probably others) use the enclosure plugin is for the active chamber heating ability, other than to switch a bunch of other stuff just on or off. I just chuck in a target temp and have a couple of ceramic heater in the enclosure that get it up to temp pretty quick.  Its very useful for all the big stuff. 
Cheers!
Philip

[image]

gdombiak, 2021-07-22:
[image]

Improved PWM and added support to control temp/humidity control outputs.

commit 824d177d (2021-07-22): Added support to control temp/humidity of enclosure plugin

Closes #491
closed 2021-07-22 by gdombiak