gdombiak/OctoPod #492: Widgets do not respect octoprints camera vertical/horizontal flipping

el0c, 2021-05-20:
As stated above, the image displayed within the widget does not get reoriented like it does within the app if you have it set to rotate within the Octoprint settings.

EvilsWay, 2021-11-17:
Plus 1 to this. 2 printers with identical cameras and considerations. One displays correctly one displays upside down in widget only. Launching the printer from the widget, video feed is correct. 

[image]

TimDowker, 2022-03-16:
Also probably related to this, notifications send to the Apple Watch do not respect camera rotation settings.

foorschtbar, 2024-09-06:
> Also probably related to this, notifications send to the Apple Watch do not respect camera rotation settings.
> 

Bump. Same problem here. Correct orientation in the App, but wrong in the notifications.