gdombiak/OctoPod #495: Feature Request - Grid View of Cameras

destinybonavita, 2021-06-11:
It would be awesome to be able to view the cameras of multiple printers at once in a zoom-like grid view.

gdombiak, 2021-06-27:
Great idea. Will try adding this to the next release. Thanks for sharing this idea.

Gaston

gdombiak, 2021-07-03:
Slowly coming along!!!

[image]

destinybonavita, 2021-07-03:
That looks awesome!

closed 2021-07-03 by gdombiak

OliS1975, 2021-08-28:
I read from the release notes of version 3.12 that the grid view is available now. But where can I activate it?

gdombiak, 2021-08-28:
Hi @OliS1975 ,

If your OctoPrint installation has [MultiCam plugin](https://plugins.octoprint.org/plugins/multicam/) installed and configured then you should see a new icon at the top-left of the window. See images below:
[image]

And this is how it looks when you clicked there. You can rotate the phone for better inspection. You can zoom in/out as well.
[image]

OliS1975, 2021-08-29:
Thank you for the explanation. So it's for a single octoprint instance with multiple cams. Not for all cams from different octoprint instances?

Oli

gdombiak, 2021-08-29:
Hey @OliS1975 , that is correct. I will be adding support for watching cameras in the printers dashboard for the next release. I remember there was an issue where someone posted a nice UI design for this but cannot find it. :(

Gaston

OliS1975, 2021-08-30:
Great to hear. 
In the meantime I installed the multicam plugin and added the cam stream from the second printer to the first one and vice versa. This works as described.

Regards
Oli

gdombiak, 2021-08-30:
Nice workaround!! :D

commit a5688b6e (2021-07-03): Added camera grid view for default printer

Closes #495