gdombiak/OctoPod #499: OctoPod and Ender 3 v2

bobster316, 2021-07-01:
I am trying to use Octopod app on my IPhone 11 with Ender 3 v2 and its not connecting, the only thing working is the camera. 
When I try to connect i get authentication failure when i try to scan the code on the machine. Am i doing somethinh wrong?

gdombiak, 2021-07-03:
Hi @bobster316,

I'm going to need more info to be able to help. I'm assuming that after a while you need an error message saying "Not refreshing". If you click on it, what error message do you see?

What's the exact error message you see when you get the authentication error? Maybe a screenshot will help.

Is scanning the QR code working? Scanning QR code should not produce an authentication error.

Are you using a reverse proxy? Are you doing port fowarding?

Gaston

gdombiak, 2021-08-29:
Hi @bobster316, any update on this issue?