gdombiak/OctoPod #500: Print Statuses Not Loading

destinybonavita, 2021-07-01:
I've been having this issue lately where statuses of prints aren't showing in the app, even when switching between different printers. 

The status will show on the home screen widget but that's it. 
[image]

gdombiak, 2021-07-03:
Hi @destinybonavita,

It looks like the websocket connection is failing to be established. After a while, do you see a "Not Refreshing" error at the bottom of the page? Could you click on it and share the error message you are seeing?

Are you using a reverse proxy? Port forwarding? There is another ticket (or was it in Facebook) that someone was using some router (cannot remember the brand) that was creating random connectivity issues. They discovered that changing some setting (sorry, again cannot remember what) made their router behave better with websockets and things work again. Might want to do a search here in github or in the facebook group.

Gaston

gdombiak, 2021-08-29:
hey @destinybonavita, are you still seeing this issue?

destinybonavita, 2021-08-29:
So sorry, I thought I replied. I am still experiencing this issue. I’m
connecting to my printers via OctoEverywhere on OctoPod. It works on the
same network without issue but via OctoEverywhere is when it’s flaky.

CannyGent, 2021-09-17:
@destinybonavita I had this issue (I am running a reverse proxy with nginx) and solved with the solution at [https://github.com/gdombiak/OctoPod/issues/175 ] (issue 175) by changing octoprint API setting to allow CORS

speedysk1, 2021-10-31:
I also get "Not refershing" and under that it says "The operation couldn't be completed.  (Starscream.WSError error 1.)

speedysk1, 2021-11-20:
I've been doing some testing and it has to be the app.  I just installed Kraken, and it's working just fine with the same network connection.  And again as noted I have always been able to use OctoEverywhere to connect OctoPod too just fine.  It's something about local connections that it's not handling properly.

Blackrider8342, 2022-02-05:
Same Issue here. App sys (Starscream.WSError error 1.).
Is there any kind of Workaround?

mrwertman, 2022-09-05:
I have this issue as well.

gdombiak, 2022-09-27:
Hey guys,

Are you all hitting the error 'Starscream.WSError error 1.'? If you do, then this is usually a connectivity error. I would need to learn more about your setup. Are you using a reverse proxy? Have you correctly configured it for websockets? Is the server & port reachable from the network used by the phone?

Regards,
Gaston