gdombiak/OctoPod #501: Allow for QR code or NFC tag to select current printer

tideline3d, 2021-07-09:
I'm running a small farm of printers.  They all look exactly the same except for their label (ex. A1) on the front.   Right now, I have to open up the app, tap on the top bar and find the corresponding printer name.  It would be super awesome if I could just tap an NFC tag or scan a QR code on the physical printer so I'm 100% sure I'm about to start/cancel the correct printer.

This may be possible with Shortcuts already?

gdombiak, 2021-07-09:
Hi @tideline3d,

That's a cool and useful idea. Thinking of implementing the following so let me know if this would solve this ask:
1. Page where you add/configure printers will have a new section with a button and input field to scan a QR code and save it associated to the printer
2. Information will be shared via iCloud to your other devices. Useful if you have an iPad and an iPhone or 2 iphones
3. On the main page, at the top left where it says Printers I will add a magnifier that when clicked will let you scan a QR code. If a printer with the scanned QR code was found then it will be selected and the dialog will be closed. If nothing was found then an error message saying "Printer not found" will appear.

Let me know if you had something else in mind.
Thanks,
Gaston

tideline3d, 2021-07-09:
Yep @gdombiak that's exactly what I was thinking.  What data will the QR codes need to contain, or can it be gibberish as long as it scans and is unique per-printer?

I'm thinking how I'd deploy the settings out for this across the farm.  I almost wonder if you shouldn't just generate a GUID that you provide to the user and say 'put this GUID into a printed QR code with some online tool'.   That way its a bit more defined what the user is supposed to do for generating QR codes.

tideline3d, 2021-07-09:
Another nice-to-have would be some sanity checks.   For example when canceling a print, require a scan of the QR code to verify which machine you're canceling.  If the QR scanned doesn't match the currently selected printer, error out.

tideline3d, 2021-07-11:
@gdombiak Another consideration for the architecture would be to just have a URL scheme registered with iOS for OctoPod then allow users to build their own QR code with that URL scheme.  Roughly something like octopod://printerID then the native QR code app you can launch from Control Center would take you into OctoPod as well.   This could likely then be leveraged to solve for NFC tags through Shortcuts + deep link URLs to the app.  

I still really like the idea of NFC taps launching me to OctoPod, much quicker when I'm moving around the farm machines constantly.  Just tap the display with my phone and instantly on the control for that printer in OctoPod

gdombiak, 2021-07-19:
@tideline3d that's a **brilliant** idea. What' even better is that it's already implemented. To use it with the existing version you need to generate a QR code with this format: ```octopod://[printerName]```.

So if your printer is called MK3 then generate this QR code: ```octopod://MK3```. I just tested it and scanning QR code from Camera opens OctoPod on the target printer. I coded this feature when I added support for widgets. 

For next release I'm going to add info to display printer URL and possibly a way to generate a QR Code so you can send it to print from the app itself.

closed 2021-07-19 by gdombiak

gdombiak, 2021-07-19:
Added new section to printer details to show iOS URL to the printer. You can press the share button to print the QR Code or save it. Did not add a scan button to change printers since Camera app can be used for this. Did you want to have a QR code scan button to change printers within the app?

[image]

tideline3d, 2021-07-19:
Nah I wouldn’t bother. Now that I know the URL stuff is implemented, I’m going to go the NFC route as it’s less swipes/taps.  I’ll test out the QR code stuff while I wait for them to ship though.   Thanks for the details!

[IMG_0316]<https://user-images.githubusercontent.com/1791267/126105767-32807f50-ec76-4fd3-b5d9-d99922ee864f.PNG>

tideline3d, 2021-07-19:
Confirmed working, FYI.  This is sweet!!

tideline3d, 2021-07-19:
@gdombiak  A couple minor bugs in this that I've found so far, let me know if you'd rather I open them separately (or just get over them :) ).  

1. Opening via QR/URL doesn't seem to load the video (an HLS stream in my case).  It loads up with a gray screen (not a connection error), navigating to another printer then back to the original target will get the image to show.  Can't seem to get it to work via QR.   A minor annoyance since I'm likely standing in front of the printer when I do this, mostly just makes me not sure it grabbed the right printer
2. If the app is on the printer menu/dashboard when you run the QR code, it will leave you in that menu screen after scanning and still show the older printer ID in the bread crumbs but when you cancel out, It goes to the newly scanned printer.  I'd expect that to just take me off the dashboard screen to the new printer I just scanned.

gdombiak, 2021-07-20:
I just fixed #474 that was causing issue 1 (almost sure). I will check out the second issue later and see what I can do about it.

Thanks so much for bringing up these issues. They help polish the app for everyone!
Gaston

a7hybnj2, 2021-07-21:
I just read through this issue as it was linked to #474 and amazing! This is such a good idea even for people with just more than one printer and not a full farm.

I was messing with the "Core NFC" framework the other day and it would be super to even include a button from the printers page that would write the url to a tag. Would be a nice user friendly way to make sure everything lined up correctly.

Thanks!

commit 37d23aff (2021-07-19): Printer Details now shows QR Code with iOS URL to the printer

Closes #501