gdombiak/OctoPod #503: Enclosure temperature/humidity not showing

webdeck, 2021-07-21:
I have the Enclosure plugin installed and the temperature and humidity values show in OctoPrint, but not in OctoPod.

OctoPrint:

[image]

OctoPod:

[image]

Linux octopi 5.10.17-v7+ #1421 SMP Thu May 27 13:59:01 BST 2021 armv7l
OctoPrint 1.6.1
OctoPod plugin 0.3.2
OctoPod iOS App v3.11 build 2

gdombiak, 2021-07-21:
Hi @webdeck,

Are you using latest version of Enclosure plugin? If not could you update and let me know if that solved it. Assuming you are using latest version ... I tried a lot to reproduce this and was not able to reproduce it. :( So I will ask for some more information to see if we can figure out what is going on.

Could you enable logging of temperature by first enabling **Enable debug info**, then save and then enable **Enable temperature debug info**. It will look like this:
[image]

After that you will need to enable debug log like this:
[image]

Last 2 steps:
1. Do you see any error in enclosure plugin logs or OctoPrint logs?
2. Could you paste the debug information that is now printing? You should see things like this

```
2021-07-20 23:06:20,080 - octoprint.plugins.enclosure - DEBUG - Sensor Input 1 Temperature: 39.5 humidity 39.5
2021-07-20 23:06:30,087 - octoprint.plugins.enclosure - DEBUG - Sensor Input 1 Temperature: 40.0 humidity 40.0
```

One more test, if you kill OctoPod app and open it again, does it consistently not show enclosure temp?

Thanks,
Gaston

webdeck, 2021-07-22:
Thanks for the debugging tips.  I was able to determine that the adafruit_dht.py script isn't working.  It was working at one point, which is probably why the web page is showing temp and humidity values, but they are old.  I'm not sure why it stopped working, but it sounds like a hardware issue.  When I run a test program, I get this error:

RuntimeError: DHT sensor not found, check wiring

closed 2021-07-22 by webdeck

gdombiak, 2021-07-22:
Yep, sounds like a hardware issue. Good that logs helped you find the root cause.

Good luck,
Gaston

webdeck, 2021-07-23:
To close the loop on this - it was a hardware issue.  I swapped out the DHT22 sensor and it is working agin.  Thanks again.