gdombiak/OctoPod #504: LabelAndFieldViewCell class missing from files, but referenced in code.

jboulter11, 2021-07-23:
```
error: Build input file cannot be found: '/Users/thunderfury/code/OctoPod/OctoPod/Panel UI/Subpanels/UI Components/LabelAndFieldViewCell.swift' (in target 'OctoPod' from project 'OctoPod')
```

This file was added to the xcproject file in this commit: https://github.com/gdombiak/OctoPod/commit/824d177decba9f58b0b91b350d413bbb5ae40553

But the file itself wasn't committed.

gdombiak, 2021-07-23:
Hey @jboulter11,

Sorry about that. For some reason xcode missed committing that file. BTW, could you please help check that your account does not have these 2 identifiers **org.OctoPod.OctoPod.watchkitapp** and or **org.OctoPod.OctoPod.OctoPod-Notification-Extension** registered in your account? This link https://developer.apple.com/account/resources/identifiers/list will show if you have them. Let me know if you do since that is blocking releasing next version of OctoPod.

Thanks,
Gaston

closed 2021-07-23 by gdombiak

jboulter11, 2021-07-23:
As far as I can tell I do not have them.  My account is not a paid developer account anymore (for at least two years now), so it is telling me I don't even have access to that page.

gdombiak, 2021-07-23:
Thanks Jim for checking. Still trying to find who owns them. Apple is trying to see what happened. Until then no new releases will come out. 😢

jboulter11, 2021-07-23:
No problem.  If they find that (somehow) I am involved, please let me know and I'll take any necessary action to release the identifiers. :)

I also checked xcode signing and it's showing I haven't even signed in to my account there, let alone selected my personal team or anything, so I doubt it's me.

gdombiak, 2021-07-23:
Agreed. Thanks again. 

Gastón

commit 09788d8d (2021-07-23): Added uncommitted file

Closes #504
Issue #491