gdombiak/OctoPod #505: Display Orange Pi temperature

jon-daemon, 2021-07-30:
Is it possible to display the SoC temp from an Orange Pi?
I use [this](https://github.com/hashashin/OctoPrint-OpiTemp) plugin to show it on the navbar.
I see that 3.11 displays rpi temp #442 but I cannot see the orange pi temp anywhere.
Also it would be awesome to see the temperature without opening the app. Maybe with a widget shortcut.

gdombiak, 2021-07-30:
Hi @jon-daemon,

I do not have an OrangePi so hard for me to test things. Couple of questions that will help:
1. Which OS and version are you running on the OrangePi?
2. Have you changed the "Custom file to read temperature from" setting of your plugin?
3. Have you installed OctoPod plugin for OctoPrint?
4. If you installed OctoPod plugin for OctoPrint, could you enable debug logging of OctoPod plugin and share logs? Plugin has support for armbian (if that is your OS). 

Thanks,
Gaston

jon-daemon, 2021-08-01:
1. I'm running armbian
2. I don't think I have changed it. It's set at ```/etc/armbianmonitor/datasources/soctemp```
3. Yes I have.
4. [Here](https://pastebin.com/SZ8VZFKU) is a recent log. I can see the temp in the log. Where should I see the temp in Octopod app?

gdombiak, 2021-08-01:
Not at a computer right now. Open Octopod and go to the second pane where you see bed and hotend temp chart. Click on the button at the top right and you’ll switch to see soc temp chart. Click on that button again to switch between both temp charts. Let me know if you see temps. Gastón

jon-daemon, 2021-08-01:
Thank you. I can see it. I thought it would be shown on the top right below hotend and heatbed temp. Sorry for the false alarm.

closed 2021-08-01 by jon-daemon

gdombiak, 2021-08-01:
That's good feedback @jon-daemon. Maybe I should also display the temp in the main panel. Happy that you can now see the SoC temp chart.