gdombiak/OctoPod #508: Chinese simplified（简体中文）Translation

clementatt, 2021-08-09:
Hi! I can help to make the Chinese simplified translation.

gdombiak, 2021-08-09:
Hi @clementatt,

That's great to hear! Attached you will find the XLIFF file with all entries that need translation. Once you are done with that, I will share about 7 more entries from OctoPod plugin that will also need translation.

[zh-Hans.xliff.zip](https://github.com/gdombiak/OctoPod/files/6955860/zh-Hans.xliff.zip)

Thanks so much!!!
Gaston

clementatt, 2021-08-12:
Hi, I finished the translation.The XLIFF file is attached below

[zh-Hans.zip](https://github.com/gdombiak/OctoPod/files/6972560/zh-Hans.zip)

> 嗨@clementatt，
> 
> 听到这个消息我很高兴！随附您将找到包含所有需要翻译的条目的 XLIFF 文件。完成后，我将分享来自 OctoPod 插件的另外 7 个条目，这些条目也需要翻译。
> 
> [zh-Hans.xliff.zip](https://github.com/gdombiak/OctoPod/files/6955860/zh-Hans.xliff.zip)
> 
> 非常感谢！！！
> 加斯顿

clementatt, 2021-08-12:
Hi, I finished the translation.The XLIFF file is attached below

[zh-Hans.zip](https://github.com/gdombiak/OctoPod/files/6972571/zh-Hans.zip)

> Hi @clementatt,
>
> That's great to hear! Attached you will find the XLIFF file with all entries that need translation. Once you are done with that, I will share about 7 more entries from OctoPod [...]

gdombiak, 2021-08-24:
Added your translation and testing it now. What is your full name so I can add you to the release notes and thanks file. :)

gdombiak, 2021-08-24:
Excellent work!! All screens I tested were translated!!! 3 questions for you:
1. What is your full name so I can add you to the release notes and thanks file. :)
2. Do you have an email account so I can share a google Excel file were I put new translations and all translators update with their translations?
3. Could you translate the following entries that belong to OctoPod plugin?
    1. 'Print complete',
    2. "Progress {}%",
    3. 'Printer bed below specified temperature threshold',
    4. 'Printer bed warmed to specified temperature and duration',
    5. 'MMU Requires User Assistance',
    6. 'Printer paused for user',
    7. 'Extruder below specified temperature threshold',
    8. 'Extruder warmed to specified temperature',
    9. 'Error {} occurred on Palette 2. Your print has been paused',
    10. 'Layer {}',
    11. 'SoC Temperature above threshold {}°',
    12. 'DANGER: Possible thermal runaway detected!'

Thanks,
Gaston

closed 2021-08-24 by gdombiak

clementatt, 2021-08-25:
Hi,You can use my Chinese name **零更酱** and my Email is **baojixlz@hotmail.com**.

The translation of the OctoPod plugin is attached below

> '打印完成',
“打印 {}％”，
'打印机床低于指定温度阈值',
'打印机床加热到指定的温度和持续时间'，
'MMU 需要用户协助',
'打印机为用户暂停',
'挤出机低于指定温度阈值'，
'挤出机加热到指定温度'，
'错误 {} 发生在Palette 2 上。您的打印已暂停',
'{} 层'，
'SoC 温度高于阈值 {}°'，
“危险：检测到可能的热失控！”

> Excellent work!! All screens I tested were translated!!! 3 questions for you:
>
> 1. What is your full name so I can add you to the release notes and thanks file. :)
> 2. Do you have an email account so [...]

gdombiak, 2021-08-25:
Thanks!!! Translation [added](https://github.com/gdombiak/OctoPrint-OctoPod/issues/111) to plugin and [thanks file](https://github.com/gdombiak/OctoPod/blob/master/THANKS.md) has been updated. :)

You should have received an email with the invite to edit the Google Excel file. Let me know if that worked fine. I added a new worksheet for Simplified Chinese. There is nothing new to translate for now. I will let you know once we have new things. If you have facebook, you can join our user group for translators. https://www.facebook.com/groups/octopod

Thanks again,
Gaston

clementatt, 2021-08-25:
I have received your email, I do not have a Facebook account, but I can register for one.
Thank you very much for developing this application. It has provided me with a lot of help to use the printer when I am not at home. I will also recommend your application in the Chinese community.
Thanks again

gdombiak, 2021-08-29:
Excellent. I just updated it and will be included in next release. Thanks!!!

clementatt, 2021-10-14:
Hello, in recent use. I found some translation errors. I have fixed them. The file is attached below.
I'm learning Swift recently, and I want to try to compile apps that include language packs by myself so that I can better check translation errors.
[zh-Hans.zip](https://github.com/gdombiak/OctoPod/files/7342869/zh-Hans.zip)
.

gdombiak, 2021-10-14:
Hi @clementatt,

I processed the update that you just shared. Could you help translate these new entries?
- Sponsors
- Become a sponsor
- Alert
- Install OctoPod plugin for OctoPrint to receive layer notifications

If you want I can guide you on how to run OctoPod in XCode and use simulator to see your translations.

谢谢
Gaston

clementatt, 2021-10-14:
Thank you very much！

gdombiak, 2022-01-01:
Hi @clementatt ,

Not sure how to reach out to you so trying here. :) Could you help translate these 2 new entries?
1. Failed to get list of relays
2. Failed to request to switch relay

谢谢
Gaston

clementatt, 2022-01-02:
1.无法获取中继列表
2.请求切换中继失败

Recently, I modified some of the uncomfortable translations based on some feedback from the Chinese 3D printing community. The file is attached below.
You can contact me with clementatt@icloud.com. Recently, Google and Microsoft's email services have network problems in China, and I cannot receive emails in time.
[zh-Hans.zip](https://github.com/gdombiak/OctoPod/files/7798469/zh-Hans.zip)

gdombiak, 2022-01-03:
Thanks for the translations. I also processed your zip file and found 12 updated translations. I think I got them all.

谢谢
Gaston

gdombiak, 2022-01-30:
Hi @clementatt,

Could you help translated the following things?

1. No image is available
2. Select printer from The Spaghetti Detective.
3. The Detective Is Not Watching
4. The Spaghetti Detective 
5. Learn More      

谢谢
Gaston

clementatt, 2022-01-30:
@gdombiak 

1. 没有可用的图像
2. 从 The Spaghetti Detective 中选择打印机。
3. 侦探没有在监测打印
4. 意大利面侦探
5. 了解更多

Please change the translation of "Complications Content" to "复杂功能内容", Apple recently modified the Chinese name of this function.

thanks.

gdombiak, 2022-01-30:
Excellent. All your changes were incorporated including the updated
translation. Thanks so much for updating the Excel file!!!!

Take care,
Gaston

gdombiak, 2022-09-26:
Hi @clementatt ,

Could you help translate this entry?

1. OctoPrint had an error using Global API Key. Remove and add printer again using 'Application Key' as a workaround

Thanks,
Gaston

clementatt, 2022-09-26:
> Hi @clementatt ,
> 
> 
> 
> Could you help translate this entry?
> 
> 
> 
> 1. OctoPrint had an error using Global API Key. Remove and add printer again using 'Application Key' as a workaround
> 
> 
> 
> Thanks,
> 
> Gaston

OctoPrint在使用全局API密钥时出错。可以使用“应用程序密钥”作为变通办法再次删除和添加打印机

Thank you very much

gdombiak, 2022-09-27:
Excellent! Thanks @clementatt for the quick translation. It has been incorporated for next release.

Gaston

gdombiak, 2023-03-20:
Hi @clementatt,

For the next release there is one new translation. Could you help translate it?

1. Install OctoPod plugin for updates

Thanks,
Gaston

clementatt, 2023-03-20:
安装OctoPod插件进行更新

gdombiak, 2023-03-20:
Wow!!! Super fast!!!! Thanks man. I just incorporated it.

Gaston

commit 74cb100e (2021-08-24): Added Simplified Chinese translation. Thanks  clementatt

Closes #508