gdombiak/OctoPod #509: OctoPod - Push Notification Issue

Honkabull, 2021-08-18:
My details are below for the setup. When the printer has an error, octopod push notification gives an option to snooze notifications for 1hr or 8hrs. Neither one worked and i got multiple notifications until i fixed the error.

- Iphone X & ios 14.7.1
- 3d Printer Prusa i3 mk3s with mmu upgrade. The error was specific to a mmu issue at the time.
- Printer is connected to Octoprint via vm & unraid via usb.
- Octopod v3.11 build 2
- Octoprint 1.6.1
- Octopod plugin 0.3.4

I was not able to find a work around and disabled push notifications during the error as it was frequently annoying and i was not nearby to fix the root issue with the printer.

gdombiak, 2021-08-20:
Hey @Honkabull,

Could you do these 2 things?
1. Enable OctoPod plugin and enable debug logging (under OctoPrint -> Settings -> OctoPod Notifications). When the error happens again, could you upload here the entire octoprint.log file? That should have the info I need to figure out what is going on
2. Could you go to OctoPrint -> Settings -> OctoPod Notifications -> Devices and confirm you do not have extra devices registered? Wonder if you are muting one but the plugin is sending notifications for the other devices (that is the same phone). Blind shot but who knows. :)

Thanks,
Gaston

Honkabull, 2021-08-20:
Gaston,

Thanks for getting back to me!

I plan to knock out a print this weekend and can for a load/unload error and provide the details you requested. Circle around tomorrow/Sunday, thanks again!

Honkabull, 2021-08-22:
Gaston,

As promised please see attached. I checked your second request and I have only 2 devices, 1 for my iPad and 1 for my iPhone.

On the 1st request, I notice I get the push notifications for MMU error based on the 5 minute interval defined in settings, screen shot attached. You can see the time stamp on my watch ~5minutes, which is great, see the pic for details. Also included the log file.

Let me know if you need anything else.

Shawn
[octoprint.log](https://github.com/gdombiak/OctoPod/files/7027661/octoprint.log)
[image]

gdombiak, 2021-08-22:
Thanks @Honkabull for sharing that info. Logs are consistent with what you are seeing (no snooze). I tested things locally and functionality should be working fine. I tested snoozing from Apple Watch and also snoozing from iPhone. 

My best guess is that the request to snooze is not getting to OctoPrint. I'm going to assume that you are using octopi. 
Could you ssh into your octopi instance and execute:
```
tail -f /var/log/haproxy.log
```
Leave that window open and cause a notification and snooze it. Send me the content you have in that window so we can confirm that OctoPod was able to request OctoPrint to stop sending requests.

Thanks,
Gaston