gdombiak/OctoPod #510: not able to delete printers or add printers from outside local sources

psdavidson812, 2021-08-29:
Hi, I am not able to delete printers or add printers from outside local sources. ip and port number is good because I can open it in my browser. Thanks

gdombiak, 2021-08-29:
Hey @psdavidson812 , are you still seeing this issue? What do you mean by local sources? Is it your local network?

psdavidson812, 2021-08-29:
Yes, I am not able to delete my printer nor am I able to edit printers.

gdombiak, 2021-08-29:
Could you check under Settings -> Security that read-only is disabled?

psdavidson812, 2021-08-29:
Bingo that was it!!!    Thanks that’s been driving me crazy!!

gdombiak, 2021-08-29:
Excellent! Another person just run into the same issue. I will improve the UI to give a hint about what is going on so users can troubleshoot this by themselves. Thanks for helping troubleshoot this.

Gaston

closed 2021-08-29 by gdombiak