gdombiak/OctoPod #517: Unable to save printer detalis

3DStyle, 2021-09-21:
Hello,
I try to edit a printer detalis (I need to chagne its IP address) and I can't click on the "Save" button - it's grayed out.
Under hostname section, it says "Enter a valid URL", altough the URL is correct. 

[image]

I tried to delete the app and reinstalled, didn't help.

gdombiak, 2021-09-23:
Hmm, I wonder why it says that URL is not valid. Does it have an empty space at the end? I will try it locally and see if I can reproduce it

gdombiak, 2021-09-23:
ok. I was able to reproduce it by doing the following:
1. Enter an invalid URL
2. Click on Scan and select an octoprint instance
3. Even though URL is completed .... message of invalid URL remains

If you tap on the URL and then tap on the printer name or API key or any other field it will remove the invalid URL message and will enable the button. I will fix this for the next release.

Thanks for reporting this issue.
Gaston

closed 2021-10-04 by gdombiak

3DStyle, 2022-08-29:
Hello,
Did this issue got fixed? I’m still having this bug on v3.23 (build 1).

commit ea157146 (2021-10-04): Clean up invalid URL error and enable save button after scanning is done

Closes #517