gdombiak/OctoPod #518: No longer connects to printer after iOS 15 update

MeganOhalloran, 2021-09-22:
As soon as I downloaded the new iOS, octopod stopped connecting to my 3b+

MeganOhalloran, 2021-09-22:
I get a message saying “Connection Failed. Service is down or incorrect port”

gdombiak, 2021-09-23:
Interesting. I just started to download iOS 15 and will test once upgrade is completed. In the meantime, basic troubleshooting ideas:
* Restart iPhone and see if that still happens
* In iOS, go to Settings -> OctoPod and make sure that **Local Network** is enabled

Let me know if any of those 2 ideas worked.
Gaston

MeganOhalloran, 2021-09-23:
Woah! Restarting worked! Local network was already enabled

gdombiak, 2021-09-23:
That's great to hear that it's now working! Restarting phone has its magic. :)

closed 2021-09-23 by gdombiak