gdombiak/OctoPod #519: Api key field does not show up

stretch-fit, 2021-10-01:
Attempting to connect to an instance on LAN, no field for an API KEY shows up on the app.

Using the "Scan" feature does find the instance, still no API key field though.

gdombiak, 2021-10-01:
Hi @stretch-fit ,

When adding a printer (OctoPrint instance) you have 3 options to choose from. You selected the first one that requires the user to press the **Request** button at the bottom of the page. OctoPod will then take you to OctoPrint webUI so you can accept the request to obtain an App Key for OctoPod.

If you want to manually enter the Global API Key then select the 3rd option. That option being deprecated by OctoPrint for their next major release. The recommended way is to use 1st option which is what you have and you will need to press Request and accept the request from OctoPrint to be able to save the new printer.

Gaston

stretch-fit, 2021-10-01:
Hi Gaston,

First,thank you so much for the response & support!

I tried that (hitting the request button at the bottom) and logged in, it took me to my instance interface through the browser but I did not see any prompt to approve for and App key request, should I be seeing something?

gdombiak, 2021-10-03:
Hi @stretch-fit ,

Yes, OctoPrint should be displaying a yellow prompt at the top right so you can approve or deny the request. You can try again and see if this time OctoPrint displays this confirmation dialog.

Worst case, use the 3rd option when adding a printer and you will be able to scan/enter the Global API Key.

Let me know how it goes.
Gaston

stretch-fit, 2021-10-04:
Interesting, no yellow prompt shows up for me on top right after logging in. I wonder if this is due to a plug-in conflict or me using Firefox as the default browser on iOS? 

I have attached a picture for what I see after logging in to for the application key request, this is after about 2-3 min of waiting.

In any case the old method with the API key appears to be working.[image]

gdombiak, 2021-10-04:
Hi Stretch-fit,

That's weird. I think that posting this in [OctoPrint's forum](https://community.octoprint.org/) or [bug tracker](https://github.com/OctoPrint/OctoPrint/issues) might be a good idea. This is how it should look like:
[image]

Regards,
Gaston

gdombiak, 2021-10-04:
After you clicked Request .... do you see any red error message? I guess you are seeing that the request timed out.

Gaston

gdombiak, 2021-10-04:
yes, you seem to have a heavily customized UI. I wonder if that is the reason for the dialog not appearing.

stretch-fit, 2021-10-04:
Hi Gaston,

I see the red timed out exception when I click back after nothing pops up. 

I bet it is due to customization of the UI, I will create a test user this evening and disable all the UI plugins and see if I can't find which one is throwing it for a loop and report back.

Stretch-fit

gdombiak, 2021-10-04:
Sounds like a good plan. Let me know what you found.

Thanks,
Gaston

stretch-fit, 2021-10-06:
Hi Gaston,

Alright so I went down the line through plug-ins to see which was preventing the pop up during application keys request. Oddly enough the only plugin that seemed to be preventing it was the TP-Link Smart Plug Control plugin. Upon disabling that plug-in the notification appears, even with all the other UI modifications.[image]

John

stretch-fit, 2021-10-06:
Not quite out of the woodwork yet, when I click "allow" on the pop-up notification it does not appear to generate a key...

[image]

gdombiak, 2021-10-10:
Hi @stretch-fit ,

Wanted to confirm that there is nothing to be fixed in OctoPod and we can close this ticket. Let me know if that is not the case.

Thanks,
Gaston

stretch-fit, 2021-10-10:
Agree, I think we are all good to close the issue. I appreciate the support!

gdombiak, 2021-10-10:
Excellent. Thanks @stretch-fit