gdombiak/OctoPod #520: Tasmota plugs not visible

FrankX0, 2021-10-03:
OctoPod is not showing my Tasmota plug.
Details:

- OctoPod: v3.13 build 1
- iOS: 15.0
- OctoPrint: 1.6.1
- OctoPod Plugin: 0.3.10 
- Tasmota-MQTT: 0.3.8

gdombiak, 2021-10-03:
Hi @FrankX0,

Have you installed [Tasmota plugin](https://plugins.octoprint.org/plugins/tasmota/)? 

Gaston

FrankX0, 2021-10-03:
Hi Gaston,

I have Tasmota-MQTT plugin installed.
Do I need Tasmota plugin instead/additionally?

Thanks,
Frank.

gdombiak, 2021-10-03:
Hey @FrankX0 ,

Tasmota-MQTT is another plugin (that is not supported by OctoPod). If you need that plugin then try adding Tasmota plugin next to it so you can control switches from OctoPod. I think that you should be able to have both installed at the same time.

Let me know how it goes.

Gaston

FrankX0, 2021-10-03:
Hi Gaston,

Yes, it's working with Tasmota directly.
I would have preferred through MQTT, but in this way it also works.
Thanks for the support.

closed 2021-10-03 by FrankX0