gdombiak/OctoPod #525: App crashes

pereiranr, 2021-10-12:
App crashes when attempting to go to files panel. 
This issue started after updating to octoprint 1.7.0

ericsnis, 2021-10-13:
I have also been experiencing crashes on OctoPod for iOS after updating Octoprint to 1.7.  The crash occurs when attempting to view the files list.  The app seems to show the camera feed just fine.

I tried the following
Uninstalling then reinstalling OctoPod from the iPhone.
Uninstall with purge data of the OctoPod plugin in Octoprint. Reinstalled again from the plugins browser in Octoprint.
Delete the printer from the Settings > Printers menu.  Then adding the printer again.
Reset iCloud data from printers under the Settings > Devices menu.

The device is a iPhone 11 running iOS 15.0.2.  OctoPod is version 3.13 build 1.

Is there a log on OctoPod that I can send?

Magnumforce40FR, 2021-10-13:
Hi
I have same issue as ericsnis after octoprint update to 1.7.
I also tried reinstalling Octopod version 3.13 build 1 from iPhone Xs running iOS 15.0.2.
I dont find any Settings -> Privacy -> Analytics & Improvements

++

gdombiak, 2021-10-14:
Guys, it would be great if you can help test the new beta to confirm the app crash is now gone. Send me an email to gdombiak at gmail if you would like to test the new beta and confirm this issue is resolved.

Thanks,
Gaston