gdombiak/OctoPod #544: Thermal runaway warning when print paused

TheCopernicus, 2021-12-02:
When the print is paused, I get warnings about possible thermal runaway. Since octopod knows it is paused and the nozzle cools down when paused, I would think it should expect to see the nozzle temp drop and not send thermal runaway notifications.

graelo, 2021-12-06:
I have the exact same remark for the heat bed. 

Thanks Gaston, your app is truly great!

gdombiak, 2021-12-06:
Thanks guys! I've been buried under work for the last few weeks and had little time to work on OctoPod. I will try to fix this issue when I have some free time.

If this error is a big inconvenience, you can disable this alert by setting "Max Temperature difference" to 0. It will disable thermal runaway monitoring. Once I fix this, you can set that value back.

Regards,
Gaston

TheCopernicus, 2021-12-07:
> Thanks guys! I've been buried under work for the last few weeks and had little time to work on OctoPod. I will try to fix this issue when I have some free time.
>
> If this error is a big [...]

No problem, totally understand. Just wanted to get it added to the list as I’ve just recently done 50+ prints with filament swaps! And octopod has been awesome!

gdombiak, 2022-01-01:
Guys, I released a new version of OctoPod plugin with a fix for this. I tested it locally and things worked fine for me. Let me know how the new version is working for you.

Thanks,
Gaston

closed 2022-01-01 by gdombiak

graelo, 2022-01-01:
Thanks Gaston! I'll try it soon. Should we install from the master branch to test it?

My octoprint indicates me the latest published version of the plugin in 3.12, although, 3.16 dates back to October. Does it mean you usually have to declare manually the version bump in octoprint plugin's repository?

gdombiak, 2022-01-01:
Hi @graelo ,

Plugin 0.3.12 is the version with the new fix. If you have it installed then you are good to go. If not update your plugin to that version. There are no changes required to iOS app.

Let me know how it goes.

Regards,
Gaston

graelo, 2022-01-03:
Hi Gaston, after upgrading the plugin to 0.3.12 I did not change my settings, but unfortunately I still get these thermal runaway notifications:

[image]

Take your time, and thanks for your efforts. Cheers!