gdombiak/OctoPod #545: Unable to login to OctoEverywhere via Google

jdtaormi, 2021-12-11:
iPhone 12 Pro 
iOS 15.1.1

Adding an OctoEverywhere printer and trying to login via Google results in

"Error 403: disallowed_useragent"

https://developers.google.com/identity/protocols/oauth2/javascript-implicit-flow#authorization-errors-disallowed-useragent

jbreu, 2022-01-01:
Yup, same for me. Can’t login and use the app because of that

gdombiak, 2022-01-01:
Hey guys,

IIRC, there is a workaround to this problem that was shared by OctoEverywhere developer. I will take a look at this error again but so far I was not able to figure out how to fix it. In the meantime, use the workaround (sorry I do not remember it - maybe it was just using regular login with your email but not sure which password - OctoEverywhere developer can help with this).

Regards,
Gaston

jdtaormi, 2022-01-01:
I believe I was able to get around it by doing a password reset for my
email which allowed me to login directly to the site

commit 1d343a5f (2022-01-30): 1. Use SFAuthenticationSession instead of WKWebView to integrate with OctoEverywhere (saved passwords can be reused and google authentication will work). 2. Generate API key via OctoEverywhere

Closes #545
Closes #558
closed 2022-01-30 by gdombiak