gdombiak/OctoPod #552: OctoPod using 100+ CPU

Nexinity7, 2022-01-20:
Using OctoPod on my iOS device fully pings the CPU on my Pi 3, however a browser instance barely uses any whatsoever, any troubleshooting tips? or is this a bug?

gdombiak, 2022-01-23:
Hi @Nexinity7,

Wow! I'm very intrigued by this finding. First time I hear about it so we will need to dive into your setup to see what is causing this and find a way to fix it.

Found this [interesting link](https://community.octoprint.org/t/high-network-usage/12221/13) that also talks about high CPU usage that seems to be coming from camera streaming. One simple test to do is to configure OctoPrint to not have a camera. Not the fix but will help narrow down the source for this so we can see how to fix it.

Could you ssh into your OctoPi instance and run top? Could you share a screenshot of what top shows?

Not likely but we can try disabling all plugins (or running in safe mode) and see what happens. Idea based on [this](https://community.octoprint.org/t/octopi-high-cpu-usage-browser-side/3525/32).

Let me know what we learned from those tests.

Thanks,
Gaston

Nexinity7, 2022-01-23:
Here it is with the app currently open[image]

gdombiak, 2022-01-23:
Top shows that the octoprint service is coming all the CPU so I do not think this is camera related. Have you tried restarting OctoPrint in safe mode and see if this still happens?

Gaston

gdombiak, 2022-01-23:
Here is another idea. In OctoPrint go to Settings -> Logging and set **tornado.access** to **INFO** level and then save the new setting. Open OctoPod for let's say 30 seconds and then close OctoPod. Attach here the octoprint.log file that will include network traffic to see if OctoPod is doing something crazy.

After this test, you can remove the logging level so your octoprint.log file stays small and clean.

Gaston

Nexinity7, 2022-01-24:
Restarting in safe mode gave me an error in OCtoPod saying the printer was not connected to octoeverywhere, so i rebooted out of it and here is the log after following the instructions you advised for tornado.access

[octoprint.log](https://github.com/gdombiak/OctoPod/files/7922247/octoprint.log)

gdombiak, 2022-01-24:
@Nexinity7 , was the CPU at 100% when OctoPrint was running in safe mode? I see only 1 warning when in safe mode but that should not case high cpu usage I think.

> 2022-01-23 21:19:29,015 - zeroconf - WARNING - Error with socket 5 (('127.0.0.1', 5353))): [Errno 22] Invalid argument
Traceback (most recent call last):
  File "/usr/lib/python3.7/asyncio/selector_events.py", line 987, in sendto
    self._sock.sendto(data, addr)
OSError: [Errno 22] Invalid argument

After the safe mode I see that you reenabled things and I see LOTS of this error which could cause high CPU usage. However, those errors do not get triggered after OctoPod or any other client connected. Might be an autoconnect and you have detect serial port? You could set the port to use rather than autoconnect. 

> 2022-01-23 21:17:23,793 - octoprint.util.comm - WARNING - Received an error from the printer's firmware: Serial status mismatch
| Last lines in terminal:
| Changing monitoring state from "Offline" to "Detecting serial connection"
| Performing autodetection with 7 port/baudrate candidates: /dev/ttyUSB0@115200, /dev/ttyUSB0@250000, /dev/ttyUSB0@230400, /dev/ttyUSB0@57600, /dev/ttyUSB0@38400, /dev/ttyUSB0@19200, /dev/ttyUSB0@9600
| Trying port /dev/ttyUSB0, baudrate 115200
| Connecting to port /dev/ttyUSB0, baudrate 115200
| Handshake attempt #1 with timeout 2.0s
| Connected to: Serial<id=0x6e297710, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=2.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
| Send: N0 M110 N0*125
| Recv: smatch
| Recv: Error:Serial status mismatch
| Recv: Error:Serial status mismatch
| Recv: Error:Serial status mismatch

After safe mode was removed, I do not see any client connecting. Hard to tell from the logs at what time you saw high CPU. If what I shared does not help narrow down the cause, then I would need to know at what time you saw high CPU so I can correlate that in the logs. Maybe check the time on the server, do your test and then send me the file again?

Gaston

Nexinity7, 2022-01-24:
CPU was at 100% for a few minutes when it was just booting into safe mode, and when i connected with OctoPod CPU went up to 90+ but never crossed 100, i'll try what you suggested in a little bit

Nexinity7, 2022-01-24:
Heres some extra info in a pastebin: https://pastebin.com/MLbsAHH1

A screenshot with top and the octoprint.log are attached and are captured when i connect with OctoPod and the CPU goes to 100
[image]
[octoprint (1).log](https://github.com/gdombiak/OctoPod/files/7928794/octoprint.1.log)
%

gdombiak, 2022-01-30:
I do not see anything particular in the webcam log that could indicate high CPU usage. I cannot see your FPS setting and if you are using any compression. I run with with 10 frames per second and compression 75%. If you have more frames per second then might want to try lowering them to see if there is any difference. However, CPU usage is in OctoPrint itself so this should not make a difference.

Looking at OctoPrint logs I see nothing suspicious that triggered high CPU from OctoPod activity. OctoPod is not flooding the server with requests or asking the server to do anything in particular. One thing that worth trying is disabling the OctoEverywhere plugin to see if that makes a difference.  That is the only thing I see running by looking at the logs. If that was not the cause for the high CPU then I would recommend posting in OctoPrint's forum. They might be able to better troubleshoot what is going on inside of OctoPrint.

Let me know how it goes.
Gaston

Nexinity7, 2022-02-03:
It had to do something with octoeverywhere; i have ran out of the trial so i wont pursue this issue until i get another chance with octoeverywhere. i'll keep you updated!