gdombiak/OctoPod #553: Disconnecting App stops running print.

stitzebr, 2022-01-22:
Rebooting your iPad shouldn’t stop a print that was never started via the app.
Poor design flaw.
I’m uninstalling till that issue is fixed.

gdombiak, 2022-01-23:
Hi @stitzebr,

OctoPod or any other client cannot stop a print by closing the app or rebooting your iPhone or iPad. I do not know what happened in your case so cannot provide much help. :( The only way to cancel a print from OctoPod or any client is to 1) press Cancel button (and you should see a warning dialog) or 2) by clicking Disconnect which will disconnect OctoPrint from the printer.

Maybe you did #2 thinking that you were disconnecting OctoPod?

Regards,
Gaston

closed 2022-01-30 by gdombiak

rebelresistor, 2022-04-22:
I ran into this problem as well today, @gdombiak . Clicking "Disconnect" doesn't do what a newbie expects it to. 

After googling, I've found numerous issues that have been opened for this problem. It seems everyone mistakes what the "Disconnect" button does. Everyone expects it to disconnect OctoPod from OctoPrint, nobody expects it to disconnect OctoPrint from the Printer. Even the confirmation dialog text doesn't trigger alarm bells in my head that this will disconnect OctoPrint from the printer.

The reason why people are having this issue, is that nobody ever clicks the "Disconnect" button in the "Connection" section of the OctoPrint Web UI, but they click "Disconnect" in apps talking to web-servers all the time. The default idea in everyone's head is that disconnecting is harmless. 

It's great that this function exists, but it needs to be buried in a sub-menu, in the Advanced settings, or something! It should definitely not be front and centre on main screen of the app, when pressing it can explode a nuclear bomb in the middle of your print. 

Outside of this, great app! Many thanks for your hard work!!