gdombiak/OctoPod #554: Synology reverse proxy

00ihsan, 2022-01-23:
I get a ssl error when trying to connect over reverse proxy. I have literally tried (almost) everything. Please help and ask what info you need.

gdombiak, 2022-01-23:
Hi @00ihsan,

I would need more info about your set up and error to be able to help with this. If the cert is self-signed then you can open OctoPod and go to Settings -> Security and disable certificate validation.

HTH,
Gaston

gdombiak, 2022-01-23:
Hmm, letsencrypt certs should be accepted by iOS. Have you tried disabling cert validation as I suggested as a workaround? Let me know if that workaround works. Your connection will still use SSL so it will be encrypted. The only downside of this workaround is what is known as man-in-the-middle attack where a hacker posses as your server and somehow was able to redirect traffic to go through their server before reaching to your printer. IMHO, low profit for such level of effort so extremely low chances of this happening.

Regards,
Gaston

00ihsan, 2022-01-23:
Hi,

I have already tried that.

gdombiak, 2022-01-23:
@00ihsan,

Would you mind if I try connecting to your server so I can debug this issue? I will be able to see exactly what is going on and be able to help. If you are ok with this, you can reach out to me via Facebook or my gmail account at gdombiak.

Gaston

00ihsan, 2022-01-23:
Hi Gaston,

I have a print running. 

Here is the API key:

C7D7B47EB8A2404E9A8ADA840B3FAD50

How long do you need it?

print.ihsansoft.nl

Met vriendelijke groet,
Ihsan Topcu

gdombiak, 2022-01-30:
It was a setup issue so closing ticket. Ping me if you need any help with this.

Gaston

closed 2022-01-30 by gdombiak