gdombiak/OctoPod #561: iOS - Unable to connect to OctoEverywhere with Google authentication

felipechaulet, 2022-02-03:
When I try to connect to OctoEverywhere and authenticate with google, I receive a message as the attached picture.

To replicate the issue:
- go to settings
- add a new printer
- select OctoEverywhere
- click on "I already have an account"
- click on "Sign in with Google"
- enter the google credentials
- you will be presented with the error in the print 

[image]

gdombiak, 2022-02-03:
Good news is that this is a duplicated of #545 which is already resolved and being released in the next release. I will see if I can submit the new version to the App Store tonight with the hope that it will be approved and go live by the weekend. In the meantime, you are welcome to give the beta version a try.

felipechaulet, 2022-02-03:
My apologies, I searched before posting the issue and couldn't find anything related to this issue.

I will close this one.

Thanks for the prompt response.
Great news that the fix is coming up soon.

closed 2022-02-03 by felipechaulet