gdombiak/OctoPod #562: Missing pause/cancel print buttons

vordenken, 2022-02-06:
The buttons are missing in the app but after force closing and reopening it the buttons appear. 
Here are two screenshots showing the problem: 

1. Buttons missing
[image]

2. Buttons showing up
[image]

Latest OctoPi, OctoPrint and OctoPod (App and Plug-in) 

If you need more informations let me know :)

gdombiak, 2022-02-07:
Hi @vordenken,

Thanks for reporting this bug. I never saw it before. Buttons should always appear when file name being printed appears. Is there a way you can provide information on how to reproduce this error? When did it start to happen? Does it happen with any file being printed? Any info would help know how to reproduce this.

Danke!
Gaston

vordenken, 2022-02-10:
Hi @gdombiak,

I think the problem only occurs if the print is already going and I open the app afterwards. If the app is open while starting the print, it shows the buttons correctly. 

It also happens with any file, tested a few yesterday. If there are any logs I can provide, let me know where to find them :) 

Gern 👍🏼

gdombiak, 2022-03-13:
Gern, I spent more time trying to reproduce this and so far no luck. Let me know if you find more info on how to reproduce this. I will ask in the Facebook group to see if anyone else is having this issue so I can figure out how to reproduce it ... and then fix it.

Danke
Gaston

PalatinCoder, 2022-03-14:
Hi @gdombiak ,
Just want to let you know that I noticed this problem as well. As @vordenken did, I too started the print via OctoPrint‘s web interface and opened the OctoPod App afterwards. Force-closing and reopening brought the buttons back. I‘ll see if I can reproduce it reliably

jimwhitelaw, 2022-04-10:
I’m able to reproduce this reliably by scrolling the bottom print info section upward until the pause/cancel buttons are hidden. When scrolling back down the buttons (text) are missing. 

On iPad, if I leave the app and return to it, they reappear. On iPhone, I can only get them back if I kill the app and restart. 

iPad Pro 11” iOS 15.3.1 
IPhone XR iOS 15.3.1

gdombiak, 2022-04-11:
Hi @jimwhitelaw,

This is great. Thanks for sharing this extra info on how to reproduce this. I will give it a try and hope to be able to reproduce this so I can fix it.

Thanks,
Gaston

gdombiak, 2022-04-11:
Thanks @jimwhitelaw . With your information I was able to reproduce this issue in no time. So weird that I never came across this bug. I will try to make room this week and fix this issue and push a new release.

Thanks again,
Gaston

jimwhitelaw, 2022-04-11:
Fantastic. I'm glad the info was helpful. The app is great and I'm very happy with it. thx.

closed 2022-04-13 by gdombiak

gdombiak, 2022-04-14:
New release is out. Let me know if you still see this issue.

Thanks,
Gaston

commit 36efc19d (2022-04-13): Scrolling down in main panel will still show pause/cancel buttons.

Fixes #562