gdombiak/OctoPod #576: Feature request: support Filament manager plugin

bushibot, 2022-03-28:
Like to request a support of filament manager.
Essential function of allowing it to notify user when not enough filament to complete job. Also be nice to be able to change the filament from the app, though less essential.

gdombiak, 2022-04-13:
Hi @bushibot ,

Filamente Manager plugin is already supported. When you open the app, you can swipe the bottom pane to the left/right to access more functionality. From there you can see remaining length for each filament and also change filaments. Getting a notification is a cool idea. That is something to add to the plugin (not the iOS app). I will move this ticket to the other repo if that is fine with you.

Thanks,
Gaston

bushibot, 2022-04-13:
Right, totally did not find that, thanks. Be nice if not buried.