gdombiak/OctoPod #577: No camera feed on my iphone

patduppat, 2022-04-12:
Octoprint run on my computer with win 10.
Cam work end on the PC
However, the application does not show the video. I also have the same video problem if I am with another pc on the same home network

gdombiak, 2022-04-13:
Hi @patduppat ,

What is the error that OctoPod displays? That should help troubleshoot the problem.

Gaston

tenumars, 2022-05-07:
I have the same error, it is unsupported url. It works with other apps and the spaghetti detective. URL is /webcam/?action=stream I don't see anything wrong, and it works with other apps just fine.

gdombiak, 2022-05-08:
Hi @tenumars,

Could you do the following and share results:
1. Check if there is an empty space before the beginning of /webcam... If there is one, try removing it and see if it works. I could fix this in next release if that is your case
2. Share a screenshot of the error you see in OctoPod? Also try clicking on the error message so that Safari opens up and share a screenshot of this as well

Thanks,
Gaston

tenumars, 2022-05-08:
Yup, there was a space. Thanks! That fixed it!

gdombiak, 2022-05-08:
Excellent! Thanks for confirming that. I will fix this for the next release so others do not have to manually remove any extra space.

Thanks,
Gaston

tenumars, 2022-05-08:
I believe it was actually caused by me, because I needed to remove the space in the octoprint ui, not the octopod one.

commit d37772a1 (2022-05-08): Remove extra white spaces at beginning/end in stream URL

Closes #577
closed 2022-05-08 by gdombiak