gdombiak/OctoPod #581: iOS app crashes when opened

IchSchreiGleich, 2022-04-21:
I can unfortunately not contribute more to the description

A few weeks ago I had set up the app and tried to get it running. I had (probably) played around with http auth. Anyway, the app no longer starts. I have already reinstalled it, but no improvement. In the icloud or via finder I can't find any files belonging to OctoPod that i could delete.

Does anyone have any ideas? I would like to avoid a reset of my phone.

Versions:
iOS 15.4.1
OctoPod latest (just downloaded today)

gdombiak, 2022-04-21:
Hi @IchSchreiGleich ,

Sorry to hear this is happening. I doubt that resetting the phone will solve this. Let me check if I can see any reported app crash that can explain what is going on so I can quickly fix it for you. BTW, do you have "Share with App Developers" enabled under Settings -> Privacy -> Analytics & Improvements? By enabling that, app developers receive more detail about app crashes that help fix things.

Thanks,
Gaston

IchSchreiGleich, 2022-04-22:
Hello @gdombiak ,
thanks for your quick reply. I got the app working and would like to share the solution here.

1. Turn off all network connections (mobile: flight mode)
2. Start the app and delete the printer that is causing the problem (menu: settings -> printer)
3. Reset the iCloud printer data (menu: settings -> devices)
4. Deactivate iCloud synchronization (menu: settings -> devices)
5. Done, now the network can be enabled again

I had to do this several times and especially **very quickly**, possibly my order was wrong and I should have turned off the synchronization first. 😉

But finally I was able to start the app and set up my printer again. The cause was probably a printer url with authorization data like `https://user:pass@printer`, but I can't say that with certainty.

What I did not like was that I had no way to delete the data manually. I could not find data via "iMazing" in the app, nor anywhere in iCloud. Maybe you could change the storage here, so that a reset of the whole app is a little bit easier. 😋

Kind regards
Marco

gdombiak, 2023-11-21:
Hi Marco,

In the last release I fixed https://github.com/gdombiak/OctoPod/issues/666 that I suspect could have been the cause for your issue. I will close this ticket for now and we can reopen it if this was not the case.

Thanks,
Gaston

closed 2023-11-21 by gdombiak