gdombiak/OctoPod #591: Timelapses with brackets in file name won’t play

theHalo, 2022-06-23:
File names of timelapse videos which contain brackets „()“ can’t be played back in the Octopod app. The screen just stays black with a loading wheel forever. 
File export and deleting works fine. 
The exported video files can be played back in the iOS Photos app. 

Maybe it’s a problem with the encoding of the file name. 

Screen recording shows me selecting 2 files with brackets, then two without and then again two with brackets and two without. 

[Screen recording ](https://share.icloud.com/photos/044Ktla0CXTz_Wo25ZZdKuKbQ)

gdombiak, 2022-06-24:
Hi @theHalo , thanks for filing this issue. I will work on it this weekend and push a new release as soon as it's fixed. Let me know if you run into anything else or have improvement ideas.

Thanks,
Gaston

closed 2022-06-25 by gdombiak

theHalo, 2022-07-16:
Just had my first timelapse with brackets in the file name and now it plays back just fine. Thanks! 👍🏼

commit 8302cc82 (2022-06-25): Fixed double URL encoding that was preventing timelapse from rendering

Closes #591