gdombiak/OctoPod #592: Octopod initiated print job by self

dbrosy, 2022-06-26:
I sent a print job to octoprint from cura 5. The job finished and 10 seconds later it restarted the print job from OctoPod

I didn't initiate this and only used octopod during print so see updates.

OctoPrint logs show print completing and then a new connection from my phone which initiated the print. 

```
2022-06-26 18:57:39,446 - octoprint.plugins.action_command_notification - INFO - Got a notification: TM3D 10SProV2 Ready.
2022-06-26 18:57:39,449 - octoprint.util.comm - INFO - Finished in 8648.736 s.
2022-06-26 18:57:39,449 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Finishing"
2022-06-26 18:57:39,476 - octoprint.printer.standard.job - INFO - Print job done - origin: local, path: CCR10SPRO_PBS_PLUS_MOUNT v5.gcode, owner: damian
2022-06-26 18:57:39,607 - octoprint.plugins.action_command_notification - INFO - Got a notification: 100% L=276/276
2022-06-26 18:57:39,640 - octoprint.util.comm - INFO - Changing monitoring state from "Finishing" to "Operational"
2022-06-26 18:57:40,694 - octoprint.plugins.DisplayLayerProgress - INFO - Printing stopped. Detailed progress stopped.
2022-06-26 18:57:40,734 - octoprint.plugins.action_command_notification - INFO - Got a notification: 0% L=276/276
2022-06-26 18:57:40,814 - octoprint.plugins.detailedprogress - INFO - Printing stopped. Detailed progress stopped.
2022-06-26 18:57:40,852 - octoprint.plugins.action_command_notification - INFO - Got a notification: Print Done
2022-06-26 18:57:50,186 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.75.117
2022-06-26 18:57:51,347 - octoprint.printer.standard.job - INFO - Print job selected - origin: local, path: CCR10SPRO_PBS_PLUS_MOUNT v5.gcode, owner: admin, user: admin
2022-06-26 18:57:51,356 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Starting"
2022-06-26 18:57:51,358 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: CCR10SPRO_PBS_PLUS_MOUNT v5.gcode, owner: admin, user: admin
2022-06-26 18:57:51,387 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2022-06-26 18:57:51,401 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"
```

192.168.75.117 = my phone

Is there a way to not allow new print jobs unless confirmed on printer?
I was not using my phone at time first print job printed

The only thing that might be relevant was I has to change the ip address of printer on OctoPod during print to be able to connect. That was at the 50% mark and I only used it to check progress

I ended up damaging the print head when it hit my first print on the bed.

Any advice would be appreciated

gdombiak, 2022-06-26:
Hi @dbrosy ,

I'm so sorry to hear the print head got damaged. That's a big no no!!! :(

The only way for OctoPod to initiate a print is to click on the print button either from the main page or the files page. If you go to **Settings** in OctoPod and click on **Prompts** you can enable **Confirm to print** to make sure that any unwanted print does not go through.

Thanks,
Gaston

gdombiak, 2022-06-26:
A confirmation dialog will appear every time you try to print from OctoPod.

closed 2022-06-28 by gdombiak