gdombiak/OctoPod #596: Not able to install from the Slovak region App store

frizzy, 2022-07-24:
Hi,

Is it possible to make this app available in the Slovakia region. I cannot switch to another region's app store to download this.

Thanks

gdombiak, 2022-07-24:
HI @frizzy,

I have not put any restrictions of where the app should be available. I expected for it to be available anywhere in the world. Let me read more about this and see if I discover anything that I could do to solve this.

Thanks for letting me know,
Gaston

gdombiak, 2022-07-24:
Yep, looking at the App Store configuration I see that it should be available everywhere in the world including Slovakia. I wonder if this is transitory issue in the App Store?

[image]

frizzy, 2022-07-25:
Hi @gdombiak .Thanks for your help! .. I managed to install it now. It appears to have been something with my iPad/Account. I'm able to install the app now. Sorry about the false alarm.

closed 2022-07-25 by frizzy