gdombiak/OctoPod #605: Time to finish printing and name of model not showing up after Octoprint update

Printyprinty, 2022-09-21:
Please help. Title and time to finish not displaying after new octoprint uodate

theHalo, 2022-09-21:
Same problem here. 
All print job information is missing. 
Temperature settings are shown but temperature  history graph says no history available. 

[image]

gdombiak, 2022-09-22:
Thanks for reporting this. I will take a look during the weekend.

Thanks,
Gaston

gdombiak, 2022-09-25:
Got to the bottom of this,. OctoPrint 1.8.3 hits an error when using global API keys as reported [here](https://github.com/OctoPrint/OctoPrint/issues/4648). I changed the API key to use an Application Key and things work well now. If it's easier, you can remove the printer and add it again to OctoPod. Make sure to select "OctoPrint - Application Key" when adding the printer again.

Let me know how it goes.

Gaston

theHalo, 2022-09-25:
Replacing the API key with a new application key worked like a charm. 
Thanks!

closed 2022-09-27 by gdombiak