gdombiak/OctoPod #606: Print - Pause - Cancel missing

mollywaters, 2022-09-22:
The ability to pause and resume or cancel a print is no longer available. These options were on the panel screen and they are now missing after a recent update.

Printyprinty, 2022-09-22:
Yeah having same problem. Hopefully the developer addresses these issues soon. Pretty annoying to have to run back to my laptop to control prints. Defeats the whole purpose of the app.

gdombiak, 2022-09-22:
Hey guys,

Sorry this is happening. Which version of OctoPrint are you running? Did this happen after you updated OctoPrint?

Thanks,
Gaston

Printyprinty, 2022-09-22:
The most recent update of 1.8.3. It have it on two of my machines and they don’t display the name, time left, cancel, or pause buttons. On the two that I didn’t upgrade, Octopod works just fine.

mollywaters, 2022-09-23:
I also have two machines also using the most recent version of Octoprint 1.8.3 both with the same issue. Sent from my iPhone

gdombiak, 2022-09-25:
Got to the bottom of this,. OctoPrint 1.8.3 hits an error when using global API keys as reported [here](https://github.com/OctoPrint/OctoPrint/issues/4648). I changed the API key to use an Application Key and things work well now. If it's easier, you can remove the printer and add it again to OctoPod. Make sure to select "OctoPrint - Application Key" when adding the printer again.

Let me know how it goes.

Gaston

mollywaters, 2022-09-25:
That worked. Thanks. Sent from my iPhone

closed 2022-09-27 by gdombiak