gdombiak/OctoPod #607: Needs updating for OctoPrint 1.8.3

megatech1966, 2022-09-23:
Needs updating for OctoPrint 1.8.3
No eta percentage etc info displayed

gdombiak, 2022-09-23:
Thanks for reporting this. I see many issues being reported after updating OctoPrint. Will take a look this weekend. Hope to have a new release soon after.

Regards,
Gaston

gdombiak, 2022-09-25:
Got to the bottom of this,. OctoPrint 1.8.3 hits an error when using global API keys as reported [here](https://github.com/OctoPrint/OctoPrint/issues/4648). I changed the API key to use an Application Key and things work well now. If it's easier, you can remove the printer and add it again to OctoPod. Make sure to select "OctoPrint - Application Key" when adding the printer again.

Let me know how it goes.

Gaston

closed 2022-09-27 by gdombiak