gdombiak/OctoPod #608: IOS 16 Widgets

samuellogan, 2022-09-25:
It would be amazing if this app added support for the new lock screen widgets on IOS 16. This would allow users to see their print progress at a glance, just like what is implemented for apple watch.

gdombiak, 2022-09-27:
Thanks @samuellogan for this request. I will try adding support for this in the 3.25 release. I'm now pushing out 3.24 as an "urgent" release.

F3000, 2022-12-24:
can you consider also live activities? 

@gdombiak

gdombiak, 2023-03-12:
Live Activities are now implemented. Starting final round of testing before calling it done. I will keep this ticket opened for just the new widgets that can be added in the lock screen /focus window.

[image]
[image]
[image]

gdombiak, 2023-03-19:
Added Lock Screen Widgets right on time before the next release!!!

[image]

closed 2023-03-20 by gdombiak